abratchik/HIDPowerDevice #2: Adapting to ATtiny85 (Digispark) and V-USB to get around CDC?

ZetaPhoenix, 2021-09-15:
I love the concept of this project and it saves me from needing to duplicate the effort but I was wondering if you have considered using an ATtiny85 board (like a digispark) as a minimal solution with V-USB?

I think this may get around the CDC issue until the other MR's get resolved and up-streamed and into device code (like in a NAS).

abratchik, 2021-09-15:
Hi @ZetaPhoenix , thanks for your interest! In theory this code should work on any Arduino Leonardo or Pro Micro clones out of the box but I never tested it on anything else. May be it could be adapted to ATtiny85, IDK. The key requirement is to be able to program HID descriptors exposed through the USB and have analog sensors or serial port pins so that the board could monitor a real power device such as UPS.

ZetaPhoenix, 2021-09-16:
```
In file included from /Users/user/Documents/Arduino/libraries/HIDPowerDevice/src/HIDPowerDevice.h:29:0,
                 from /Users/user/Documents/Arduino/UPS/UPS.ino:1:
/Users/user/Documents/Arduino/libraries/HIDPowerDevice/src/HID/HID.h:26:10: fatal error: PluggableUSB.h: No such file or directory
 #include <PluggableUSB.h>
          ^~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board Digispark.
```
Looks like some more work will be required to port but I am seeing limitations on the IO pins if you want to monitor charge input and have more status LED's so the Arduino Pro Micro may be the way to go.

korenevskiy, 2021-12-24:
I bought an ATiny 85 Digispark for power control to make USB UPS.

The fact is that I already have a working UPS. But the feature of the UPS I bought does not have a built-in USB(COM) port. But I don't need to make changes to this device. I need a completely different device. I want to have an adapter that connects to the USB port of a PC, while the adapter must have an additional separate 5v power connector. I will connect a charger from a cell phone to this connector, while the charger will work from the mains. As soon as the electricity goes out, the adapter will count down 5 minutes and then send a signal to the PC to switch to hibernation mode.
Optionally, digispark could control the voltage of 12V. But this is not necessary. I want to make an adapter with a minimum of effort in the electrical circuit. Simply connect the power of an external charger from a cell phone to digispark.

korenevskiy, 2021-12-24:
What is the CDC problem?