abratchik/HIDPowerDevice #10: HID Battery not UPS

hellboysatan, 2024-08-05:
Would it be possible to refactor that program to show it's the battery and not the UPS?

abratchik, 2024-08-05:
Sorry, did not understand. The way how it is shown in the OS is controlled by the OS. Usually the UPS is shown as the battery.

hellboysatan, 2024-08-05:
I would like it to not show the UPS name, but only the battery, or the Microsoft AC adapter if possible.

abratchik, 2024-08-06:
not to show where? The device name is constructed by the host OS. Usually it takes the device type, which is taken from the HID constant and may be PID/VID.

forderud, 2024-10-14:
I suspect that what is really being asked here is how to change the HID descriptor so that the device appears like a "Battery" instead of a "UPS" on connected computers. I'm also interested in this myself.

The project currently appears like a "HID UPS Battery" on Windows, and similarly a "UPS" on Linux and macOS. It would be nice if it could also be configured to instead show up as a "Battery".
[image]

I've already tried to change `0x09, 0x04, // USAGE (UPS)` on the top of the HID descriptor in [HIDPowerDevice.cpp](https://github.com/abratchik/HIDPowerDevice/blob/master/src/HIDPowerDevice.cpp) to either ` 0x09, 0x10, // USAGE (BatterySystem)
` or  `0x09, 0x12, // USAGE (Battery)`. `BatterySystem` and `Battery` are documented with "Physical collection" usage type in the standard, whereas `UPS` is documented with usage type "Application collection". The `0xA1, 0x01, // COLLECTION (Application)` line below therefore also needs to be modified to `0xA1, 0x00, // COLLECTION (Physical)`.

However, only changing the top-level USAGE to Battery(System) and COLLECTION to Physical only leads to Windows no longer recognizing the device as a UPS or Battery. I therefore suspect that more invasive changes to the HID descriptor are needed to make the device appear as a "Battery".

@abratchik Any suggestions or comments?

forderud, 2024-10-14:
Status update:
I've now inspected `hidbatt.inf`, which is the in-built driver used when testing on Windows:  
[image]

The INF file hardcodes `HID_DEVICE_BATTERY.DeviceDesc = "HID UPS Battery"`, which is probably the source of the device name. I'm not sure though if this hardcoding is fixed, or it it's possible to override it with special HID descriptors.

hellboysatan, 2025-01-23:
@forderud I want to ask if you found out how to change it?

forderud, 2025-01-23:
> [@forderud](https://github.com/forderud) I want to ask if you found out how to change it?

I've unfortunately not been able to change how the battery appear in Widows.