abratchik/HIDPowerDevice #11: Windows: Device report 0% charge if additional batteries already exist

forderud, 2024-08-28:
First, thanks a lot for sharing this project!

I noticed that the Arduino device always shows up as a battery with 0% charge when testing in Windows on a laptop that already has a battery.
[image]

## Work-around
It appear to be possible to work around this problem by disabling the real battery device in Device Manager:
[image]
The remaining Arduino battery then afterwards shows up with correct charge:  
[image]

It would be great if the device could show up with correct charge also on multi-battery setups.

abratchik, 2024-08-28:
You are welcome. This is known issue but the problem, unfortunately. 

Windows 10 does allow multiple batteries in the system and all of them can be displayed in the tray. However, the batteries must report their status in the same mode (HID specification supports maH, mwH and percentage) but most of UPS are only reporting percentage of charge. If you already have ACPI battery in your system (let's say you are testing on a laptop) then the percentage may not be reflected correctly. 

You may change the units in the sketch how the remaining charge is reported and this will fix the issue but it will be valid for Windows host with the ACPI battery, such as the laptop, for example.

forderud, 2024-08-29:
> You may change the units in the sketch how the remaining charge is reported and this will fix the issue but it will be valid only for the Windows host with the ACPI battery.

I just tested changing `HID_PD_CAPACITYMODE` from 2 (%) to 1 (mwH) and it did indeed fix the problem. Thanks a lot for the tip! I fully understand that all values here are artificial, so whether or not the aggregated battery capacity % for all batteries reported by Windows is correct or not doesn't really matter for me. I'm just interested in making the capacity always show up on screen.

Are you interested in updating the `UPS.ino` example with this change? If so, then I can submit a PR th

abratchik, 2024-08-29:
Well, if your fix is tested on Linux, Mac  and Windows then of course I will accept, why not? :-) Will be happy to.

forderud, 2024-08-29:
> Well, if your fix is tested on Linux, Mac and Windows then of course I will accept, why not? :-) Will be happy to.

Great! However, it will take me some days to test on all those platforms. Will get back to you when I'm done.

forderud, 2024-08-29:
See #13 for a proposed fix this this annoyance.